cmbahadir/opcua-esp32 #1: Function undeclared error when compiling

houfengmz, 2018-07-17:
Hi, I got this error below, it seems that the function 'mbedtls_platform_entropy_poll' will be defined only on Unix and Windows, is there anything wrong?

>D:/msys32/home/hf/esp/opcua-esp32/components/open62541lib/open62541.c:48528:42: error: 'mbedtls_platform_entropy_poll' undeclared (first use in this function)
                                          mbedtls_platform_entropy_poll, NULL, 0,
                                          ^
D:/msys32/home/hf/esp/opcua-esp32/components/open62541lib/open62541.c:48528:42: note: each undeclared identifier is reported only once for each function it appears in
D:/msys32/home/hf/esp/opcua-esp32/components/open62541lib/open62541.c: In function 'policyContext_newContext_sp_basic256sha256':
D:/msys32/home/hf/esp/opcua-esp32/components/open62541lib/open62541.c:49517:42: error: 'mbedtls_platform_entropy_poll' undeclared (first use in this function)
                                          mbedtls_platform_entropy_poll, NULL, 0,
                                          ^
make[1]: *** [/home/hf/esp/esp-idf/make/component_wrapper.mk:286：open62541.o] 错误 1
make: *** [D:/msys32/home/hf/esp/esp-idf/make/project.mk:468：component-open62541lib-build] 错误 2
>

In esp-idf/components/mbedtls/mbedtls/library/entropy_poll.c
```
#if !defined(MBEDTLS_NO_PLATFORM_ENTROPY)

#if !defined(unix) && !defined(__unix__) && !defined(__unix) && \
    !defined(__APPLE__) && !defined(_WIN32) && !defined(__QNXNTO__)
#error "Platform entropy sources only work on Unix and Windows, see MBEDTLS_NO_PLATFORM_ENTROPY in config.h"
#endif
````

cmbahadir, 2018-07-17:
Hi hufenggmz,

1 - Thanks for your feedback, have you faced with below error before entropy_poll;

root@galliumos:~/esp/esp-idf/examples/open62541# make
CC build/nodeset/simple.o
/home/cmb/esp/esp-idf/examples/open62541/components/nodeset/simple.c: In function 'function_simple_6_begin':
/home/cmb/esp/esp-idf/examples/open62541/components/nodeset/simple.c:236:11: error: implicit declaration of function 'UA_encodeBinary' [-Werror=implicit-function-declaration]
 retVal |= UA_encodeBinary(&variablenode_ns_1_i_6006_Argument_0_0_struct.Name, &UA_TYPES[UA_TYPES_STRING], &posvariablenode_ns_1_i_6006_Argument_0_0, &endvariablenode_ns_1_i_6006_Argument_0_0, 
           ^
cc1: some warnings being treated as errors
/home/cmb/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'simple.o' failed
make[1]: *** [simple.o] Error 1
/home/cmb/esp/esp-idf/make/project.mk:468: recipe f

2- And also which OS you work on?

houfengmz, 2018-07-18:
1. Yes, The header file 'ua_types_encoding_binary.h' amalgamated into the single source file lead to this warning, so I changed the build options to ignore this warning temporarily.
2. FreeRTOS

cmbahadir, 2018-07-18:
Hi Houfenggmz,

I made few modifications to open62541 as below, which i haven't flashed yet but compiled successfully.
 
1 - open62541 should be built with UA_ENABLE_ENCRYPTION flag OFF (other flags should remain the same)

 2 - Few security functions comes undeclared;
    - UA_SecurityPolicy_Basic128Rsa15
    - UA_SecurityPolicy_Basic256Sha256
    - UA_encodeBinary
    - UA_decodeBinary

mbedtls and other security plugins for ESP32 board can also be raised as an issue at;
https://github.com/open62541/open62541.git

Give it  try with my latest commit and let me know if ou face anything further.

closed 2018-07-18 by cmbahadir