cmbahadir/opcua-esp32 #10: Problems with compilaton

rodrigolange, 2019-06-22:
Hi, 

First of all, congratulations for your work in OPC UA.

But I need some help. I'm trying to compile your project, but I'm having several compilation problems.

I'm using the latest esp-idf and xtensa-esp32-elf toolchains (downloaded from the official site as today), in a Ubuntu 18.04 machine, and I followed the toolchains instalation instructions in Espressif site. I'm also able to compile the examples without problems.

Regarding the compilation of opcua-esp32: initially, the compilation was aborted with the error 

/opcua-esp32/components/freertos/include/freertos/portmacro.h:81:10: fatal error: esp_crosscore_int.h: No such file or directory

As a test, I comented the line including esp_crosscore_int.h in portmacro.h file, but then the compiler returns the following error:

esp/esp-idf/components/driver/periph_ctrl.c:31:5: error: implicit declaration of function 'portENTER_CRITICAL_SAFE'; did you mean 'portENTER_CRITICAL_ISR'? [-Werror=implicit-function-declaration]
     portENTER_CRITICAL_SAFE(&periph_spinlock);

Reading the Espressif documentation, it seems that for compatibility reasons functions portENTER_CRITICAL_SAFE and portEXIT_CRITICAL_SAFE are now portENTER_CRITICAL_ISR and portEXIT_CRITICAL_ISR.

I tried to rename the functions on the files indicated by the compiler (there are lots of them) with no luck, because after it the compilation was aborted with errors regarding tcp/ip variables (I don't have the exact errors anymore).

I believe that the problem is related to the versions of the toolchains and the FreeRTOS, but I'm not sure. Can you help me?

Thanks in advance,

Rodrigo

Thanks in advance,

cmbahadir, 2019-06-23:
Hello Rodrigo,

Thanks for your interest in the project.

The problem is probably related to the esp-idf version, you are using. It has to be v3.2. 

Can you please switch it to v3.2 tag with;

git checkout tags/v3.2
git submodule update --init 

And let me know about the result.

Happy Sunday.

cmbahadir, 2019-07-11:
Closing the RQ since there is no other build problem observed.

closed 2019-07-11 by cmbahadir