cmbahadir/opcua-esp32 #14: How to Debug example code?

kusumy, 2019-10-21:
Hi @cmbahadir , can you give me an information how to debug the code?
I want to look at it in more detail.

I am using your example to read data from one-wire device. 
I changed a DHT22 device to 1-wire device.

For reading data from 1-wire device, i use this repository:
https://github.com/DavidAntliff/esp32-ds18b20-example

I run the 1-wire sample code from repository, and its working.

The strange is when i implement the code from esp32-ds18b20-example to opcaua example, the temperature always contant value (100). 

I still do not know what went wrong, so i think i need to debug it to find out.

Regards

cmbahadir, 2019-10-21:
Hi @kusumy,

Actually 100 is error temperature value which i set as a workaround and it indicates a problem with the sensor. There may be a connection or interfacing error. Here is the error check;

https://github.com/cmbahadir/opcua-esp32/blob/791565336222c3e33ae35aea071add2cd9447e08/components/DHT22/DHT22.c#L237

To debug the esp32 you need the configuration stated debugging chapter of esp-idf guide but keep in mind that an additional hardware like JTAG-Debugger is needed;

https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/using-debugger.html

Have a nice week.

kusumy, 2019-10-23:
Also when i try to write value to relay using UAExpert, there is an error message:

_Write to node 'NS1|String|Control Relay number 1.' failed [ret = BadInternalError]_

But i can set Blink value successfully.

cmbahadir, 2019-10-23:
I couldn't reproduce your problem by myself, can you please provide Wireshark logs if possible and also that would be great if you can set the log level as 100 and provide me the "make monitor" logs of your ESP32.

You should set the log level as below line in "components/open62541lib/include/open62541.h";

`#define UA_LOGLEVEL 500` to
`#define UA_LOGLEVEL 100`

kusumy, 2019-10-24:
hi @cmbahadir, i really appreciate that you share your code.
I am still new OPC UA, especially for embedded. Your code saves me.  Thank you.
If you don't mind, please explain in more detail about your code,  especially about app_main() function.
I do not see any opc ua function, and i am still confuse why you put wifiscan() function at the end, not at beginning.

btw, if i want to add other sensor/data, do i just need to put the code on model.c/h and called the code from app_main() ?

I will try to capture the wireshark log and make monitor log, and send to you as soon as possible?

Regards

kusumy, 2019-10-25:
hi @cmbahadir, below is "make monitor" log when i tried to set relay value:

[1970-01-01 00:03:46.250 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Processing ReadRequest
[1970-01-01 00:03:46.260 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Read the attribute 13
[1970-01-01 00:03:46.280 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | Sample callback called
[1970-01-01 00:03:46.290 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Read the attribute 13
[1970-01-01 00:03:46.310 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | The value has not changed
[1970-01-01 00:03:46.520 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | Sample callback called
[1970-01-01 00:03:46.520 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Read the attribute 13
[1970-01-01 00:03:46.540 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | The value has not changed
[1970-01-01 00:03:46.650 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | Publish Callback
[1970-01-01 00:03:46.770 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | Sample callback called
[1970-01-01 00:03:46.770 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Read the attribute 13
[1970-01-01 00:03:46.790 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | The value has not changed
[1970-01-01 00:03:47.020 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | Sample callback called
[1970-01-01 00:03:47.020 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Read the attribute 13
[1970-01-01 00:03:47.040 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | The value has not changed
[1970-01-01 00:03:47.150 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | Publish Callback
[1970-01-01 00:03:47.270 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | Sample callback called
[1970-01-01 00:03:47.270 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Read the attribute 13
[1970-01-01 00:03:47.290 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | The value has not changed
[1970-01-01 00:03:47.520 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | Sample callback called
[1970-01-01 00:03:47.520 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Read the attribute 13
[1970-01-01 00:03:47.540 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | The value has not changed
[1970-01-01 00:03:47.650 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | Publish Callback
[1970-01-01 00:03:47.770 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | Sample callback called
[1970-01-01 00:03:47.770 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Read the attribute 13
[1970-01-01 00:03:47.790 (UTC+0000)] debug/session	Connection 56 | SecureChannel 2 | Session ns=1;g=506acff1-52ad-8e7d-9799-39f32112d1bd | Subscription 2 | MonitoredItem 1 | The value has not changed

And here is Wireshark log:
"207","90.749975001","192.168.244.155","192.168.100.98","OpcUa","162","UA Secure Conversation Message: ReadRequest"
"208","90.750710622","192.168.100.98","192.168.244.155","TCP","60","4840  >  51186 [ACK] Seq=3066 Ack=3784 Win=64240 Len=0"
"209","90.998648287","192.168.100.98","192.168.244.155","OpcUa","128","UA Secure Conversation Message: ReadResponse"
"210","91.041012953","192.168.244.155","192.168.100.98","TCP","54","51186  >  4840 [ACK] Seq=3784 Ack=3140 Win=46320 Len=0"
"211","91.046979020","192.168.100.98","192.168.244.155","OpcUa","139","UA Secure Conversation Message: PublishResponse"
[... cut ...]
"222","95.986672874","192.168.100.98","192.168.244.155","TCP","60","4840  >  51186 [ACK] Seq=3374 Ack=4059 Win=64240 Len=0"
"223","96.007688644","192.168.244.155","192.168.100.98","OpcUa","162","UA Secure Conversation Message: ReadRequest"
"224","96.007903735","192.168.100.98","192.168.244.155","TCP","60","4840  >  51186 [ACK] Seq=3374 Ack=4167 Win=64240 Len=0"
"225","96.265169241","192.168.100.98","192.168.244.155","OpcUa","128","UA Secure Conversation Message: ReadResponse"
"226","96.308442195","192.168.244.155","192.168.100.98","TCP","54","51186  >  4840 [ACK] Seq=4167 Ack=3448 Win=46320 Len=0"

Regards

cmbahadir, 2019-11-23:
Hello @kusumy , 

> If you don't mind, please explain in more detail about your code, especially about app_main() function.
> I do not see any opc ua function, and i am still confuse why you put wifiscan() function at the end, not at beginning.

I am also not that familiar with open62541 stack itself, but order of the functions is not important, since opcua is a separate freertos task. So first it scans the WIFI with configured SSID and when connected it runs the opcua server task. 

> 
> btw, if i want to add other sensor/data, do i just need to put the code on model.c/h and called the code from app_main() ?
Yes, you can add them just like relays or temperature and call them from the opcua_esp32.c - opcua_task. But keep in mind that memory is limited and more data source means more memory.

> 
> I will try to capture the wireshark log and make monitor log, and send to you as soon as possible?
> 
I couldn't see any problem in your logs, can you please try with latest HEAD. https://github.com/cmbahadir/opcua-esp32/commit/5a57cfa42e8a2f698bd2f27ebfd5850edafb04b9

Here, i have changed the access level mask of the temperature data as read-only, because it think it is more logical to not modify it over client. Also when it was wanted to write temperature data, it was crashing. I couldn't get how did you manage to write temperature value.

kusumy, 2019-11-29:
Finally i can have your code working, after i create new environment for the code. I do not know what makes your code not working on the previous environment.

Congrats, i think you have made a cheapest OPC UA device on the planet.

cmbahadir, 2019-11-29:
Great have fun. :) But please be aware when you are using it for a real world application since its not completely tested.

closed 2019-11-29 by cmbahadir