cmbahadir/opcua-esp32 #16: Cannot connect and disconnect more than 9x

GISED-Link, 2019-12-02:
Hi,

Thank you for providing a demo code wich run quite nice. But when we connect/disconnect more than 9x times, we get "bad communication error". In wireshark, we see that the client initiate the communication correctly (TCP connection and send the hello message), but the server (or esp32?) automaticaly closed the tcp connection. 

We have made some changes to the code to be able to use the ETH interface. So maybe we have messed up something. Can you try at your side to connect/disconnect more than 10x ?

Our configuration :

- Build on a Windows 10 machine
- ESP-IDF V3.3
- https://github.com/OLIMEX/ESP32-EVB/tree/master/SOFTWARE/ETH_PHY_demo for the ETH interface
- open62541 version: Git-Revision: 0.3-rc2-1038-g2e14f900
- client: UaExpert and https://github.com/FreeOpcUa/opcua-client-gui

In our debug task, we see that the static ram is allocated when a new connection is made and released correctly (same amount of byte before before connection and after release). The task stack usage stay unchanged before and after the connection. With 10'240 bytes, the task should have enough stack. After the initialization, it has 5'000 bytes to work with.

Thank you for your work 👍

cmbahadir, 2019-12-02:
Hello @GISED-Link ,

I have checked it with both UaExpert and FreeOpcUa/opcua-client-gui and i couldn't reproduce your case. IMO your ETH configuration may caused something.

[image]

By the way current tag of open62541 of me is v1.1-dev-228-gde6f14f9 and esp-idf is v4.1-dev-1086-g93a8603c5 , which i have updated with PR #15 .

Thanks for the compliments of you, keep in mind that this repo is just a try and not optimized to work as a professional opc ua server. But any contributions are welcomed ;)

GISED-Link, 2019-12-02:
Hi @cmbahadir,

Thank you a lot for the quick reply. Knowing that you are able to run the open62541 with the latest release is a big and a greate news. To upgrade to the ESP-IDF 4.x, we need to perform changes to the ETH interface we use. So I must check with my team what is best to do now 🤔 

> not optimized to work as a professional opc ua server

I must say it does quite a good job! The discovery thing works well, we can connect with more than 1 client at the same time ... Promessing 🥇

Thank you for your wrok.

closed 2019-12-02 by GISED-Link

cmbahadir, 2019-12-02:
Just one thing to remind, after updating to latest open62541 there appeared a hard fault when setting discovery url, spent some time but couldn't find the root cause. 

So i commented it out here;

https://github.com/cmbahadir/opcua-esp32/blob/771b79f16a1995f7838fbbf3a0cbcb7247462905/main/opcua_esp32.c#L49

fyi.

GISED-Link, 2019-12-02:
Can you open an issue with this? So I can see when it will be resolved (or contribute directly)

cmbahadir, 2019-12-02:
Opened #17 

Cheers.

cmbahadir, 2019-12-06:
Hello @GISED-Link,

> I must say it does quite a good job! The discovery thing works well, we can connect with more than 1 client at the same time ... Promessing 🥇

I just  wondered about your discovery configuration. Do you have a dedicated discovery server? And also do you have an mDNS service running on ESP32?
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/mdns.html

If it works for you too i am planning to add an mDNS service to be used by OPC UA Discovery Service. But first i got to understand configuration needed :)

GISED-Link, 2019-12-12:
Sorry for the delay... 

I don't remember all the story about this. I wait on @https://github.com/MattPfenn for an answer.

What I can say right now:

1. mDNS should work but it is not the perfect solution (see 2.)
2. You can send the Hostname in the DHCP (bootp) request

For the second option, you need to change lwip/opt.h to tell lwip to send the hostname during the DHCP request.

in the few following days I will certainly update the code necessary to do this. But I need first to configure a local network with a proper dns server. Our office denied to register our esp32 boards ...