cmbahadir/opcua-esp32 #17: Setting Discovery URL Causes Hard Fault

cmbahadir, 2019-12-02:
Setting the discovery url causes crash. 

Server Code:
https://github.com/cmbahadir/opcua-esp32/blob/771b79f16a1995f7838fbbf3a0cbcb7247462905/main/opcua_esp32.c#L49

Stack Trace:

```
I (6165) esp_netif_handlers: sta ip: 192.168.1.100, mask: 255.255.0.0, gw: 192.168.0.1
I (6165) APP_MAIN: Event id: 7
I (6165) APP_MAIN: SYSTEM_EVENT_STA_GOT_IP
I (6665) APP_MAIN: After OPC UA Task : 227776
I (6665) APP_MAIN: Fire up OPC UA Server.
[1970-01-01 00:00:06.340 (UTC+0000)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | Read the attribute 2
[1970-01-01 00:00:06.350 (UTC+0000)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[1970-01-01 00:00:06.360 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[1970-01-01 00:00:06.370 (UTC+0000)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Node type ns=2;i=1002 not found
I (7045) APP_MAIN: Data Memory Heap Left : 95100
I (7045) APP_MAIN: Free Memory: 95100
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "D:/Task_Force/ESP32/esp-idf/components/heap/heap_caps.c", line 271, function: heap_caps_free
abort() was called at PC 0x40161463 on core 1
0x40161463: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: 794c40a047e6961044a7cc0c48b268b02603cc40d21017b6875378a5625b0a36

Backtrace: 0x4008f9c1:0x3ffc87e0 0x4008fd49:0x3ffc8800 0x40161463:0x3ffc8820 0x400823c7:0x3ffc8850 0x40094fa1:0x3ffc8870 0x400da6e6:0x3ffc8890 0x400daabc:0x3ffc88b0 0x400d9566:0x3ffc88d0 0x400da6d4:0x3ffc88f0 0x400fb05d:0x3ffc8910 0x400d6138:0x3ffc8990 0x40092049:0x3ffc8a90
0x4008f9c1: invoke_abort at D:/Task_Force/ESP32/esp-idf/components/esp32/panic.c:157
0x4008fd49: abort at D:/Task_Force/ESP32/esp-idf/components/esp32/panic.c:174
0x40161463: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)
0x400823c7: heap_caps_free at D:/Task_Force/ESP32/esp-idf/components/heap/heap_caps.c:271 (discriminator 1)
0x40094fa1: free at D:/Task_Force/ESP32/esp-idf/components/newlib/heap.c:47
0x400da6e6: UA_Array_delete at d:\task_force\esp32\esp-idf\examples\opcua-esp32\build/../components/open62541lib/open62541.c:9593
0x400daabc: String_clear at d:\task_force\esp32\esp-idf\examples\opcua-esp32\build/../components/open62541lib/open62541.c:8529
0x400d9566: UA_clear at d:\task_force\esp32\esp-idf\examples\opcua-esp32\build/../components/open62541lib/open62541.c:9522
0x400da6d4: UA_Array_delete at d:\task_force\esp32\esp-idf\examples\opcua-esp32\build/../components/open62541lib/open62541.c:9589 (discriminator 3)
0x400fb05d: UA_Server_run_startup at d:\task_force\esp32\esp-idf\examples\opcua-esp32\build/../components/open62541lib/open62541.c:25368
0x400d6138: opcua_task at d:\task_force\esp32\esp-idf\examples\opcua-esp32\build/../main/opcua_esp32.c:72
0x40092049: vPortTaskWrapper at D:/Task_Force/ESP32/esp-idf/components/freertos/port.c:143
```
open62541.c - Function - Line: 9593
```
void
UA_Array_delete(void *p, size_t size, const UA_DataType *type) {
    if(!type->pointerFree) {
        uintptr_t ptr = (uintptr_t)p;
        for(size_t i = 0; i < size; ++i) {
            UA_clear((void*)ptr, type);
            ptr += type->memSize;
        }
    }
    UA_free((void*)((uintptr_t)p & ~(uintptr_t)UA_EMPTY_ARRAY_SENTINEL));
}
```

cmbahadir, 2019-12-02:
Yeap here i found the reason :

https://github.com/open62541/open62541/commit/a06a792fa810bff6705df749fdc1da109aab6986#diff-6ad32046c79b2c74ce2b938c870442a9

GISED-Link, 2020-01-06:
Hi,

We currently work on it with MattPfenn. The link you gave us seems not enough to make the discovery running. 

We are working deeper on this. We use the 4.0 release from ESP-IDF and v1.1-dev-228-gde6f14f9 for the Open62541. The ETH driver works fine.

Tell me if you have already found a solution.

cmbahadir, 2020-01-06:
Hey,

Just check this out; 
https://github.com/Pro/open62541-esp32/blob/ca014bb51409b0280a0747698cd09ca355f5a066/main/main.c#L98

This is @Pro's port of open62541 to esp32. He is also one of the core developers of open62541 stack so he knows what he is doing better than us i guess :) and seems like he managed to configure discovery with mdns, i know you don't prefer the mdns way, but it may be the only choice for now.

Yeah it was my fault not to delete the comment, the root cause was not the given link, sorry for that.

>We use the 4.0 release from ESP-IDF and v1.1-dev-228-gde6f14f9 for the Open62541. The ETH driver works fine.

And great to see its working with ETH, good job well done.

I am getting prepared to FOSDEM2020 @Brussels these days so i will not able to contribute here, so don't expect any solution from me for a while :)

cmbahadir, 2020-03-08:
Currently  the only for discovery is to mdns and its implemented with latest commit. The server will be seen as "opcua-esp32" in discovery server.

So ı am closing this issue but i will track the topic.

closed 2020-03-08 by cmbahadir