cmbahadir/opcua-esp32 #25: Issue with building

zoran93, 2020-04-21:
Hi,
i have a problem with building the project.
The picture below shows the problem.
[image]

Maybe you know what could be wrong?

Best regards,

cmbahadir, 2020-04-22:
Hi,

This is an issue with esp-idf, you can track the issue thread;
[https://github.com/espressif/esp-idf/issues/5046]

A workaround would be to use an older tag of esp-idf which dates back before the issue, and make a clean build afterwards.

closed 2020-04-22 by cmbahadir

zoran93, 2020-04-22:
Thanks for your help.

Another thing, if you know from pratice. If I have one analog input, what is the maximum speed to upload the measured value to OPC-UA client? How fast is OPC-UA on esp32?

cmbahadir, 2020-04-22:
I haven't tested the minimum sampling interval, but at least 1 s works for me. :) Please let me know when you try smaller values.