cmbahadir/opcua-esp32 #27: it not work in ESP32-WROVER?

acer1204, 2020-05-30:
i download this project and compile it success and download it in ESP32-WROVER-B.
how to test it?

i open com monitor, it not show anything?

i need use idf.py menuconfig to change some config?

cmbahadir, 2020-05-30:
Is there any ethernet module on the board, and do you want to use it? 

If there is no physical ethernet configuration, the only choice is to use the WI-FI on ESP32, so you need to switch the config to use WiFi instead by changing the below line;

https://github.com/cmbahadir/opcua-esp32/blob/98c6ac288a431cfef1d5e374f283735d7d5076e2/components/ethernet/include/ethernet_connect.h#L19

to

`#define CONFIG_EXAMPLE_CONNECT_WIFI 1`

I will move these fields to sdkconfig in the future.

acer1204, 2020-06-01:
> 板上是否有任何以太網模塊，您是否要使用它？
> 
> 如果沒有物理以太網配置，唯一的選擇是使用ESP32上的WI-FI，因此您需要通過更改以下行來切換配置以使用WiFi；
> 
> https://github.com/cmbahadir/opcua-esp32/blob/98c6ac288a431cfef1d5e374f283735d7d5076e2/components/ethernet/include/ethernet_connect.h#L19
> 
> 至
> 
> `#define CONFIG_EXAMPLE_CONNECT_WIFI 1`
> 
> 將來，我會將這些字段移至sdkconfig。

just replace "CONFIG_EXAMPLE_CONNECT_ETHERNET  1"  to "CONFIG_EXAMPLE_CONNECT_WIFI 1" and no other code needs to be changed in opcua-esp32 project?

acer1204, 2020-06-01:
> Is there any ethernet module on the board, and do you want to use it?
>
> If there is no physical ethernet configuration, the only choice is to use the WI-FI on ESP32, so you need to switch the config [...]

just replace "CONFIG_EXAMPLE_CONNECT_ETHERNET 1" to "CONFIG_EXAMPLE_CONNECT_WIFI 1" and no other code needs to be changed in opcua-esp32 project?

closed 2020-06-01 by acer1204

cmbahadir, 2020-06-01:
After setting your wi-fi SSID and Password from `idf.py menuconfig`,  flash and run the `idf.py monitor`, the ip address after a succesful connection will appear on the log and you can connect from an OPC UA Client by using that IP address.

acer1204, 2020-06-02:
> After setting your wi-fi SSID and Password from `idf.py menuconfig`, flash and run the `idf.py monitor`, the ip address after a succesful connection will appear on the log and you can connect from an OPC UA Client by using that IP address.

thank you very much!
It work Now.

cmbahadir, 2020-06-02:
Your welcome, that would be great if you leave a star to the project. :)