cmbahadir/opcua-esp32 #28: Problems compiling with ESP-IDF v4.0.1

fradaxx, 2020-06-01:
Hi,
I'm attempting to compile your project using ESP-IDF v4.0.1 on a Windows 10 PC; I setup esp-idf using the esp-idf-tools-setup-2.3.exe tool and all gone well (no setup errors).

If I try to compile your project I get error

< 
../main/opcua_esp32.c:6:10: fatal error: esp_netif.h: No such file or directory
 #include "esp_netif.h"
          ^~~~~~~~~~~~~
>
in the last steps of compilation; please find attached the complete compilation log.
[compile_log.txt](https://github.com/cmbahadir/opcua-esp32/files/4709798/compile_log.txt)

In effect i can't find esp_netif folder and all related .h/.c files in components subfolder of esp-idf, but if I try to sync to the git repo with the command

<
git clone --recursive https://github.com/espressif/esp-idf.git <your-esp-idf-root-path>
>

you suggest on your guide, I get the error
<
fatal: destination path 'esp-idf' already exists and is not an empty directory.
>

Please can you help me with this problem?

Thanks in advance.

Regards

cmbahadir, 2020-06-01:
Seems like there is a problem with your local esp-idf configuration have you followed all of the steps here;

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer

fradaxx, 2020-06-01:
Thank you for your answer.

Yes, I followed step by step the instructions given by Espressif; the setup gone well and this is confirmed by the correct build of many predefined examples.

The strange thing is that, in particular, the folder esp_netif is not present at all in my setup in the components subfolder but it is present on the master branch of the ESP-IDF git repository; however I'm unable to update my local setup of ESP-IDF (4.0.1) after the installation because git gives the reported non-empty folder error.

This happens both if I choose to download the current release version of ESP-IDF (v4.0.1) directly by the Espressif setup tool and also if I download it by means of git in a local folder and then ask the setup tool to use this forlder as a source for the installation process.

Any hint will be appreciated.

cmbahadir, 2020-06-01:
You may need to run;

git submodule uptade —init —recursive

in your local esp-idf path.

Or when you clone it for the first time run it with option recursive like;

git clone —recursive <esp-idf git path>

Seems like submodules are missing on your esp-idf path.

fradaxx, 2020-06-01:
Dear cmbahadir,
I confirm that there was a problem with submodules sync in my local path; following your instructions all goes right!

I just successfully built your project and now I am trying to look for published variables using KEPServerEX!

Thank you very much for your kind assistance.

Regards

cmbahadir, 2020-06-01:
Have fun and dont forget to leave a star :)

closed 2020-06-01 by cmbahadir