cmbahadir/opcua-esp32 #29: ESP32-EVB Ethernet Connection

rony91jp, 2020-11-13:
Hello,

I tried to connect the ESP32-EVB board via LAN cable but the i am facing the following issue. Please refer to the below.  
[image]
Could you please tell me where to set the base MAC address? 

Thanks

cmbahadir, 2020-11-14:
Hi @rony91jp ,

What happens after? 
Can you please send me the full log and your configuration? 
Is it a local or a corporate network?
And lastly, does it work when the wi-fi is enabled?

rony-91, 2020-11-24:
Hi,

Sorry for the late reply. I changed the board to esp32-POE. When the wi-fi enabled, i was able to connect the board to the UaExpert but still not able to connect via LAN. Below is the full log when using the esp32-POE board. 
Using Python in C:\Python38\
Python 3.8.2
Using Git in C:\Program Files\Git\cmd\
git version 2.21.0.windows.1
Setting IDF_PATH: C:\Users\saha_ronykumer\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\saha_ronykumer\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Users\saha_ronykumer\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin
    C:\Users\saha_ronykumer\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Users\saha_ronykumer\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\saha_ronykumer\.espressif\tools\idf-exe\1.0.1\
    C:\Users\saha_ronykumer\.espressif\tools\ccache\3.7\
    C:\Users\saha_ronykumer\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\saha_ronykumer\.espressif\python_env\idf4.3_py3.8_env\Scripts

Checking if Python packages are up to date...
Python requirements from C:\Users\saha_ronykumer\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Users\saha_ronykumer\esp-idf>cd examples

C:\Users\saha_ronykumer\esp-idf\examples>cd opcua^esp32
指定されたパスが見つかりません。

C:\Users\saha_ronykumer\esp-idf\examples>cd opcua-esp32

C:\Users\saha_ronykumer\esp-idf\examples\opcua-esp32>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build
Executing "ninja all"...
[1/10] Performing build step for 'bootloader'
ninja: no work to do.
[8/8] Generating binary image from built executable
esptool.py v3.0-dev
Generated C:/Users/saha_ronykumer/esp-idf/examples/opcua-esp32/build/opcua_esp32.bin

Project build complete. To flash, run this command:
C:\Users\saha_ronykumer\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe ..\..\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\opcua_esp32.bin
or run 'idf.py -p (PORT) flash'

C:\Users\saha_ronykumer\esp-idf\examples\opcua-esp32>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build
Executing "ninja all"...
[1/10] Performing build step for 'bootloader'
ninja: no work to do.
[8/8] Generating binary image from built executable
esptool.py v3.0-dev
Generated C:/Users/saha_ronykumer/esp-idf/examples/opcua-esp32/build/opcua_esp32.bin

Project build complete. To flash, run this command:
C:\Users\saha_ronykumer\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe ..\..\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\opcua_esp32.bin
or run 'idf.py -p (PORT) flash'

C:\Users\saha_ronykumer\esp-idf\examples\opcua-esp32>idf.py -p COM4 flash
Executing action: flash
Running ninja in directory c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\saha_ronykumer\esp-idf\components\esptool_py && C...build" -P C:/Users/saha_ronykumer/esp-idf/components/esptool_py/run_esptool.cmake"
esptool.py --chip esp32 -p COM4 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 opcua_esp32.bin
[... 162 lines cut ...]
0x4008a62d: vPortTaskWrapper at C:/Users/saha_ronykumer/esp-idf/components/freertos/xtensa/port.c:170

ELF file SHA256: ff0d2b91bdef90c7

Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG esp-2020r3) 8.1.0.20180627-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build\opcua_esp32.elf...done.
Remote debugging using \\.\COM4
warning: unrecognized item "T0b" in "qSupported" response
0x4008772a in panic_abort (details=0x3ffbc6fb "abort() was called at PC 0x40086687 on core 0") at C:/Users/saha_ronykumer/esp-idf/components/esp_system/panic.c:360
360         *((int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
(gdb)

cmbahadir, 2020-11-24:
`E (711) emac_esp32: emac_esp32_init(351): reset timeout`
`E (711) esp_eth: esp_eth_driver_install(203): init mac failed`

These two lines show that your ethernet configuration is not correct.
You need to configure the ethernet parameters properly, since ESP32-PoE is a different board with different ethernet configuration.

There was a discussion about this in esp-idf issues, i guess it will solve your problem if you follow the steps here;
https://github.com/espressif/esp-idf/issues/4565

closed 2020-11-24 by cmbahadir

rony91jp, 2020-11-25:
Hello,

Sorry to re-open it. I tried to connect the ESP32-EVB board via LAN But couldn't get the IP address.  It works when using Wi-Fi and able to connect to UaExpert. Please check the below.

Using Python in C:\Python38\
Python 3.8.2
Using Git in C:\Program Files\Git\cmd\
git version 2.21.0.windows.1
Setting IDF_PATH: C:\Users\saha_ronykumer\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\saha_ronykumer\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Users\saha_ronykumer\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin
    C:\Users\saha_ronykumer\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Users\saha_ronykumer\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\saha_ronykumer\.espressif\tools\idf-exe\1.0.1\
    C:\Users\saha_ronykumer\.espressif\tools\ccache\3.7\
    C:\Users\saha_ronykumer\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\saha_ronykumer\.espressif\python_env\idf4.3_py3.8_env\Scripts

Checking if Python packages are up to date...
Python requirements from C:\Users\saha_ronykumer\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Users\saha_ronykumer\esp-idf>cd examples

C:\Users\saha_ronykumer\esp-idf\examples>cd opcua-esp32

C:\Users\saha_ronykumer\esp-idf\examples\opcua-esp32>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.

Project build complete. To flash, run this command:
C:\Users\saha_ronykumer\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe ..\..\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\opcua_esp32.bin
or run 'idf.py -p (PORT) flash'

C:\Users\saha_ronykumer\esp-idf\examples\opcua-esp32>idf.py -p COM4 flash
Executing action: flash
Running ninja in directory c:\users\saha_ronykumer\esp-idf\examples\opcua-esp32\build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\saha_ronykumer\esp-idf\co...ronykumer/esp-idf/components/esptool_py/run_esptool.cmake"
esptool.py --chip esp32 -p COM4 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 opcua_esp32.bin
esptool.py v3.0-dev
Serial port COM4
Connecting........_
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:11:ae:d8:ec:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 4195.8 kbit/s)...
Hash of data verified.
Compressed 25072 bytes to 15426...
Writing at 0x00001000... (100 %)
[... 89 lines cut ...]
I (0) cpu_start: App cpu up.
I (427) cpu_start: Pro cpu start user code
I (427) cpu_start: cpu freq: 160000000
I (427) cpu_start: Application information:
I (432) cpu_start: Project name:     opcua_esp32
I (437) cpu_start: App version:      7c45ba3-dirty
I (442) cpu_start: Compile time:     Nov 25 2020 15:37:00
I (449) cpu_start: ELF file SHA256:  6adf011a4b00c6df...
I (454) cpu_start: ESP-IDF:          v4.3-dev-1720-g494a124d9-dirty
I (462) heap_init: Initializing. RAM available for dynamic allocation:
I (469) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (475) heap_init: At 3FFB9078 len 00026F88 (155 KiB): DRAM
I (481) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (487) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (494) heap_init: At 40090CE4 len 0000F31C (60 KiB): IRAM
I (501) spi_flash: detected chip: generic
I (505) spi_flash: flash io: dio
W (509) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (523) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (612) system_api: Base MAC address is not set
I (612) system_api: read default base MAC address from EFUSE
I (1132) esp_eth.netif.glue: 4c:11:ae:d8:ec:23
I (1132) esp_eth.netif.glue: ethernet attached to netif
I (1142) online_connection: Waiting for IP

cmbahadir, 2021-04-07:
Hello @rony-91 

Unfortunately I do not have a ESP32-PoE board, have you got the chance to solve this issue. 
That would be greate if you could state what you did to get the IP address properly in an PoE board.

ZhidongYa, 2021-11-08:
Hi, @cmbahadir , 

I tried to connect the ESP32-EVB board via LAN cable and in a local network run the code from https://github.com/OLIMEX/ESP32-EVB/tree/master/SOFTWARE/ESP32-EVB_Ethernet_v4.0. when idf.py monitor, a problem comes:
E (479) lan8720: lan8720_pwrctl(357): power up timeout
E (479) lan8720: lan8720_init(419): power control failed
E (479) esp_eth: esp_eth_driver_install(205): init phy failed

hier is the flog:
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\esp\.espressif\python_env\idf4.3_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\esp\.espressif\tools\idf-git\2.30.1\cmd\
git version 2.30.1.windows.1
Setting IDF_PATH: C:\Users\zhido\Desktop\esp-idf

Adding ESP-IDF tools to PATH...
    C:\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin
    C:\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin
    C:\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin
    C:\esp\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin
    C:\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\esp\.espressif\tools\cmake\3.16.4\bin
    C:\esp\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin
    C:\esp\.espressif\tools\ninja\1.10.2\
    C:\esp\.espressif\tools\idf-exe\1.0.1\
    C:\esp\.espressif\tools\ccache\3.7\
    C:\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\zhido\Desktop\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\Users\zhido\Desktop\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Users\zhido\Desktop\esp-idf>cd examples\ESP32-EVB_Ethernet_v4.0

C:\Users\zhido\Desktop\esp-idf\examples\ESP32-EVB_Ethernet_v4.0>idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory c:\users\zhido\desktop\esp-idf\examples\esp32-evb_ethernet_v4.0\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\users\zhido\desktop\esp-idf\examples\esp32-evb_ethernet_v4.0"...
-- Found Git: C:/esp/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/zhido/Desktop/esp-idf/examples/ESP32-EVB_Ethernet_v4.0/sdkconfig
-- Found PythonInterp: C:/esp/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe (found version "3.8.7")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "ethernet_basic" version: v4.3.1
[... 437 lines cut ...]
0x4008400f: _esp_error_check_failed at C:/Users/zhido/Desktop/esp-idf/components/esp_common/src/esp_err.c:42

Backtrace:0x400850d7:0x3ffb6f20 0x400858c9:0x3ffb6f40 0x4008ad0e:0x3ffb6f60 0x4008400f:0x3ffb6fd0 0x400d5f4f:0x3ffb6ff0 0x400fde05:0x3ffb7060 0x400889d5:0x3ffb7080
0x400850d7: panic_abort at C:/Users/zhido/Desktop/esp-idf/components/esp_system/panic.c:368

0x400858c9: esp_system_abort at C:/Users/zhido/Desktop/esp-idf/components/esp_system/system_api.c:112

0x4008ad0e: abort at C:/Users/zhido/Desktop/esp-idf/components/newlib/abort.c:46

0x4008400f: _esp_error_check_failed at C:/Users/zhido/Desktop/esp-idf/components/esp_common/src/esp_err.c:42

0x400d5f4f: app_main at c:\users\zhido\desktop\esp-idf\examples\esp32-evb_ethernet_v4.0\build/../main/ethernet_example_main.c:120 (discriminator 1)

0x400fde05: main_task at C:/Users/zhido/Desktop/esp-idf/components/freertos/port/port_common.c:133 (discriminator 2)

0x400889d5: vPortTaskWrapper at C:/Users/zhido/Desktop/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: d32ab2728ba32172

Rebooting...
ets Jun  8 2016 00:22:57

About configuration:
[image]
[image]

cmbahadir, 2021-11-09:
Hello @ZhidongYa 

It seems that you are struggling with building the ethernet example from esp-idf repo. It looks similar to this issue:

https://github.com/espressif/esp-idf/issues/6977

And there is already a PR to fix this. 

That would be more appropriate to follow framework problems in https://github.com/espressif/esp-idf.

cmbahadir, 2021-11-09:
This is not a problem of opcua-esp32. Better to be addressed in https://github.com/espressif/esp-idf/issues

You can find more people there who might also have the same problem with ethernet example.