cmbahadir/opcua-esp32 #32: How to assign Static IP

rony91jp, 2020-12-01:
Hello,

I am Currently using ESP32-EVA board and able to connected to Wi-Fi as well to the UaExpert. 
My next step is to Connect this board with ethernet and want to assign a static IP address. 
I follow the instructions step by step but couldn't get the assigned ip address. 
In the ethernet_connect.h file i wrote the static ip as below. 
[image]
Output
[image]

Thanks

rony-91, 2020-12-01:
The above problem has solved.

cmbahadir, 2020-12-01:
Hello rony-91,

How did you solve it? It is not yet implemented, have you managed to set a static IP.

rony91jp, 2020-12-10:
Hello, 

Sorry for the late reply.
I just added the below and it works.

static void start(void)
{  
  
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&cfg);
    assert(eth_netif);
    s_example_esp_netif = eth_netif;

    // Set default handlers to process TCP/IP stuffs
    ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &on_got_ip, NULL));

    //Configuration using LAN8720
    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();

    phy_config.phy_addr = CONFIG_EXAMPLE_ETH_PHY_ADDR;
    phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO;
    mac_config.smi_mdc_gpio_num = CONFIG_EXAMPLE_ETH_MDC_GPIO;
    mac_config.smi_mdio_gpio_num = CONFIG_EXAMPLE_ETH_MDIO_GPIO;
    s_mac = esp_eth_mac_new_esp32(&mac_config);
    s_phy = esp_eth_phy_new_lan8720(&phy_config);
    
    esp_eth_config_t config = ETH_DEFAULT_CONFIG(s_mac, s_phy);

    tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_ETH);
    tcpip_adapter_ip_info_t ipInfo;
    IP4_ADDR(&ipInfo.ip, 192,168,3,5);
    IP4_ADDR(&ipInfo.gw, 192,168,3,1);
    IP4_ADDR(&ipInfo.netmask, 255,255,255,0);
    tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_ETH, &ipInfo);
    ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle));
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
    s_connection_name = "ETH";
}
I have an one issue. 
I am now having a problem with this following command.
UA_NodeId currentNodeId = UA_NODEID_STRING(nsIndex, "Char");
If i set nsIndex=1, It works.
When i changed the nsIndex=2 it doesn't works.
Where do i need to change?

Thank you.

cmbahadir, 2020-12-10:
Great! I am glad that you have solved your problem.

I am planning a big merge to master with new features. Including setting a static IP over a GUI tool.

closed 2020-12-10 by cmbahadir