cmbahadir/opcua-esp32 #37: need help pls

leonvita91, 2021-04-07:
hello 

thank you for your code it helps me so much and it working just fine, so my question is when i connect with wifi everything works fine and when i connect with Ethernet i got an error on the console and it tells me the esp32 POE has no mac addr !! and normally the esp Ethernet comes without any MAC addr i tried to change some code but my experience with C language not good enough,and after long research i found that i should add this mac addr manually .. may you pls help me in this point .

cmbahadir, 2021-04-07:
Hello @leonvita91

This is the same issue with #29 

E (711) emac_esp32: emac_esp32_init(351): reset timeout
E (711) esp_eth: esp_eth_driver_install(203): init mac failed

These two lines show that your ethernet configuration is not correct.
You need to configure the ethernet parameters properly, since ESP32-PoE is a different board with different ethernet configuration.

There was a discussion about this in esp-idf issues, i guess it will solve your problem if you follow the steps here;
espressif/esp-idf#4565

leonvita91, 2021-04-13:
hello again 

i tried to understand the options that i possible to solve this problem, and one says i should make the rest time to 500ms,so i started to search about where i can change the rest time but couldn't found it may you pls help i will be happy for an answer . 
init(351): reset timeout 

this is what im searching for 351ms to 500ms 

greetings leon :))

leonvita91, 2021-04-14:
hello thank you again for this nice project and i would close the issue because i solve it. 

so the solution that i changed the driver with the one worked for my board and if anyone had this board 

[image]
[image]

so what i did, i just added block by block the driver for this board ESP32-POE  in ethernet_connect.c that you can find it in this DIR >>> /…/examples/opcua-esp32/components/ethernet so this file is now customized 
and now looks like this and it work just fine for me :) 
and in here you just need this code and go to the idf.py menuconfig >> component config >> ethernet >> support ESP32 internal EMAC controller >> RMII clock mode >>> and choose Output RMII clock >> go back and chnage RMII clock GPIO number to 17 . and flash it :)) thats it !! 

/////////////////////////////////////////////////////////////////
#include <string.h>
#include "sdkconfig.h"
#include "esp_event.h"
#include "esp_wifi.h"
#include "esp_wifi_default.h"

#if CONFIG_EXAMPLE_CONNECT_ETHERNET
#include "esp_eth.h"
#endif

#include "esp_log.h"
#include "driver/gpio.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "lwip/err.h"
#include "lwip/sys.h"
#include "tcpip_adapter.h"

#include "ethernet_connect.h"

#define GOT_IPV4_BIT BIT(0)
#define GOT_IPV6_BIT BIT(1)
#define CONNECTED_BITS (GOT_IPV4_BIT)

static const char *TAG = "online_connection";

/** Event handler for Ethernet events */
static void eth_event_handler(void *arg, esp_event_base_t event_base,
                              int32_t event_id, void *event_data)
{
    uint8_t mac_addr[6] = {0};
    /* we can get the ethernet driver handle from event data */
    esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;

    switch (event_id) {
    case ETHERNET_EVENT_CONNECTED:
        esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
        ESP_LOGI(TAG, "Ethernet Link Up");
        ESP_LOGI(TAG, "Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                 mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
        break;
    case ETHERNET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "Ethernet Link Down");
        break;
    case ETHERNET_EVENT_START:
        ESP_LOGI(TAG, "Ethernet Started");
        break;
    case ETHERNET_EVENT_STOP:
        ESP_LOGI(TAG, "Ethernet Stopped");
        break;
    default:
        break;
    }
}

/** Event handler for IP_EVENT_ETH_GOT_IP */
static void got_ip_event_handler(void *arg, esp_event_base_t event_base,
[... 224 lines cut ...]
    ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_CONNECTED, &on_wifi_connect));
#endif
    esp_err_t err = esp_wifi_stop();
    if (err == ESP_ERR_WIFI_NOT_INIT) {
        return;
    }
    ESP_ERROR_CHECK(err);
    ESP_ERROR_CHECK(esp_wifi_deinit());
    ESP_ERROR_CHECK(esp_wifi_clear_default_wifi_driver_and_handlers(s_example_esp_netif));
    esp_netif_destroy(s_example_esp_netif);
    s_example_esp_netif = NULL;
}
#endif //CONFIG_EXAMPLE_CONNECT_WIFI

esp_netif_t *get_example_netif(void)
{
    return s_example_esp_netif;
}
                                                               `
///////////////////////////////////////////////////////////////////

i hope this solution help someone else :) 

greetings 
leon :+1:

closed 2021-04-14 by leonvita91