cmbahadir/opcua-esp32 #39: Max number of nodes in OPC UA

betodr4, 2021-04-23:
Hi Dear!

I am working in a project where I have to manage more than 100 variables to monitor them via OPC UA protocol with an ESP32 and I have choose your library.
I am having problems due to management of more than 20 variables. I use UA Expert as a Client and I can`t monitor more than 10 variables. With open62541 v1.1 everything woks properly but the memory consumption is not very adjust so I upgrade to version 1.2 and I started obtaining this problems. I will atached here some pictures of the errors I am obtaining related with Out of Memory.

In UA Expert I have the following logs:
- Error 'BadOutOfMemory' was eturned during OpenSecureChannel.

In ESP32 console I obtain the followinf errors and then the board just reset:
- emac esp32: no mem for receive buffer

I am trying to adjust some LWIP parameters for the protocol but I don't obtain any results.

If ypu need some information, please contact me.

Thank you very much.

Alberto Delgado.

cmbahadir, 2021-04-25:
Hello Alberto,

I can't see the attached pictures. 
I have not tested the maximum number of variables, but seems like the limit is 10 for your code, I don't think that 100 variables could be possible, it might be with ESP32-WROOM of 16 MB of flash.

Seems like there is no memory left for receive/send buffer. 

You can try decreasing `recvBufferSize` and `sendBufferSize` to expand the space for your information model. I did this by trial and error until I got the successful connection.

https://github.com/cmbahadir/opcua-esp32/blob/3f6f789ada11387961ac09c2c7bd70a927da395f/main/opcua_esp32.c#L73-L74

Does it work when you decrease the number of nodes/variables?
If you have build(amalgamated) the open62541 by yourself, which build options have you used?
It completely depends on your implementation, so if possible please send a code snippet. 
Please also provide some more logs by setting open62541 UA_LOGLEVEL to 600 (trace).

Cheers,
Mete

betodr4, 2021-04-26:
Dear Cmbahadir.

Thank you so much for your feedback.

Rightnow I don't really know how to configure the server with the 150
variables. Do you think creating one node per each variable is the optimus
solution?

The final idea is to read the variables of another side and update them
every second in the OPCUA server. I have try by the creation of one node
per each variable and the function Write to modify the value of each
variable every Sample time. I attached you my main.c and my model.c and
model.h where I have defined the structure of the function for the creation
and writing of the variables.

If you think there is another better option please let me know. I don´t
know so much in OPC UA and maybe I could be stressing too much the buffers
and the memory flash of the ESP 32.

Thank you so much.

Alberto Delgado.

#include <stdio.h>
#include <sys/param.h>
#include <unistd.h>
#include <lwip/sockets.h>
#include <esp_flash_encrypt.h>
#include "esp_netif.h"
#include <esp_task_wdt.h>
#include <esp_sntp.h>
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_event.h"
#include "nvs_flash.h"
#include "sdkconfig.h"

#include "ethernet_connect.h"
#include "open62541.h"
#include "model.h"

#define EXAMPLE_ESP_MAXIMUM_RETRY 10

#define TAG "OPCUA_ESP32"
#define SNTP_TAG "SNTP"
#define MEMORY_TAG "MEMORY"
#define ENABLE_MDNS 1

static bool obtain_time(void);
static void initialize_sntp(void);

UA_ServerConfig *config;
static UA_Boolean sntp_initialized = false;
static UA_Boolean running = true;
static UA_Boolean isServerCreated = false;
RTC_DATA_ATTR static int boot_count = 0;
static struct tm timeinfo;
static time_t now = 0;

static UA_StatusCode UA_ServerConfig_setUriName(UA_ServerConfig *uaServerConfig, const char *uri, const char *name) {
	// delete pre-initialized values
	UA_String_clear(&uaServerConfig->applicationDescription.applicationUri);
	UA_LocalizedText_clear(&uaServerConfig->applicationDescription.applicationName);

	uaServerConfig->applicationDescription.applicationUri = UA_String_fromChars(uri);
[... 345 lines cut ...]
//}
//void add_Node(UA_Server *server, char *Name) {
//	UA_VariableAttributes attr = UA_VariableAttributes_default;
//	attr.displayName = UA_LOCALIZEDTEXT("en-US", Name);
//	attr.dataType = UA_TYPES[UA_TYPES_FLOAT].typeId;
//	attr.accessLevel = UA_ACCESSLEVELMASK_READ;
//	attr.historizing = 1;
//
//	UA_NodeId currentNodeId = UA_NODEID_STRING(1, Name);
//	UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, Name);
//	UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
//	UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
//	UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0,
//	UA_NS0ID_BASEDATAVARIABLETYPE);
//
//
//	UA_DataSource Variable1;
//	Variable1.read = set;
//	UA_Server_addDataSourceVariableNode(server, currentNodeId, parentNodeId,
//			parentReferenceNodeId, currentName, variableTypeNodeId, attr,
//			Variable1, NULL, NULL);
//}
//
///*--------------------------------------------------------------------------------------*/
///* END */

cmbahadir, 2021-05-03:
Hello Alberto,

Please take a look at this section in open62541 wiki.
https://open62541.org/doc/current/tutorial_server_datasource.html
Which explains how to connect a datasource to a node variable.

It will probably consume less memory than your code but there will still be a limit for max number of variables.

leonvita91, 2021-05-26:
Hello Alberto, 
i had solved your problem in my issue so check this out 
https://github.com/cmbahadir/opcua-esp32/issues/42
greetings 
leon :)

betodr4, 2021-05-26:
Hello leonvita91 and cmbahadir.

I also solved the problem and it was easier than I expected. I just had to enable the option "Initialize SPI RAM during startup".

With this option we start using the 4MB RAM memory of the ESP32 and there are enought memory for all the nodes you want.

Thanks everyone.

aldelro ;)

cmbahadir, 2021-05-26:
Hello @leonvita91 @betodr4 

Great to see that both problems have been solved by small memory tricks.

Thanks for contribution.

I will close this issue and please dont forget to star this repo if you are using. ☺️

leonvita91, 2021-05-27:
**Hello Alberto,** 
thank your for answer 
I would say your Solution is very good for **SPI** in order to active the external memory which great,but unfortunately this option will not working with **I2C** or the Ethernet has no **SPI**. in this case you need to deal with this few heap memory to disable mdns and less nodes which is sad but i still happy with this project and im so thankful for everyone share knowledge with others :1st_place_medal: 

best regards 
**Leon**

closed 2021-05-27 by cmbahadir