cmbahadir/opcua-esp32 #42: esp.emac no mem for receive buffer

leonvita91, 2021-05-18:
helllo 

i had problem with my OPC-UA which is the mem buffer , did you had any experience with this error,also my buffer run out too fast and not clean the buffer memory ? for the information i will  upload my skconfig file  

`#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config
[... 1244 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options`

cmbahadir, 2021-05-18:
Hello Leon,

Logs help more than sdkconfig. 
Please set the `CONFIG_UA_LOGLEVEL` as `100` and provide the output of `idf.py monitor`.

Are you using any additional nodes, methods or attributes except the ones in the master branch? If yes then please provide your customized code snippet.

What kind of ESP32 development board you are using?

In the meantime you can play with the receive and send buffer sizes here;
https://github.com/cmbahadir/opcua-esp32/blob/daba7cbed3c845f26bb8cb88a1f2cbcebea4eaf7/main/opcua_esp32.c#L73-L74

leonvita91, 2021-05-19:
Thank you for answer,im using ESP32 POE ISO.
What i had done,i changed the Ethernet driver also added bme280 with DHT22  and i had delete the two relays because i don't had them on my board ,so all the information that you need is on my Repo on this link  
https://gitlab.hzdr.de/alqase03/opc-ua-with-bme280
so i did the research and i found that , if i want to use the external memory i should  using SPI which is i didn't because im using i2c not SPI so long story short im still had problem with mem buffer !_! 
greetings leon

leonvita91, 2021-05-25:
hello again i want to close this issue because i had solved the problem and i would like to share this solution with others and i had make long long search about it no solve for this problem so the funny thing after one week debugging i had found the mdns take from the heap memory and makes the heap full  so the solution is so easy you just need to deactivated the mdns thats all 
so go the opcua_esp32.c 
and search for mdns and make it false like here :) 

#ifdef ENABLE_MDNS
    config->mdnsEnabled = false;
    config->mdnsConfig.mdnsServerName = UA_String_fromChars(appUri);
    config->mdnsConfig.serverCapabilitiesSize = 2;
    UA_String *caps = (UA_String *)UA_Array_new(2, &UA_TYPES[UA_TYPES_STRING]);
    caps[0] = UA_String_fromChars("LDS");
    caps[1] = UA_String_fromChars("NA");
    config->mdnsConfig.serverCapabilities = caps;

for me it works really fine with 3 nodes 
also keep in mind more nodes more heap memory so you can control it from 
Here !!
//BufferSize's got to be decreased due to latest refactorings in open62541 v1.2rc.
    UA_Int32 sendBufferSize = 10000; // 16384;
    UA_Int32 recvBufferSize = 10000; //16384;

and here 
        xTaskCreatePinnedToCore(opcua_task, "opcua_task", 24000, NULL, 10, NULL, 0);

i will be happy if this solution helps the others :+1: 

greetings leon :)

closed 2021-05-25 by leonvita91

HyungGyuKim76, 2021-06-25:
> hello again i want to close this issue because i had solved the problem and i would like to share this solution with others and i had make long long search about it no solve for this problem so the funny thing after one week debugging i had found the mdns take from the heap memory and makes the heap full so the solution is so easy you just need to deactivated the mdns thats all
> so go the opcua_esp32.c
> and search for mdns and make it false like here :)
> 
> #ifdef ENABLE_MDNS
> config->mdnsEnabled = false;
> config->mdnsConfig.mdnsServerName = UA_String_fromChars(appUri);
> config->mdnsConfig.serverCapabilitiesSize = 2;
> UA_String *caps = (UA_String *)UA_Array_new(2, &UA_TYPES[UA_TYPES_STRING]);
> caps[0] = UA_String_fromChars("LDS");
> caps[1] = UA_String_fromChars("NA");
> config->mdnsConfig.serverCapabilities = caps;
> 
> for me it works really fine with 3 nodes
> also keep in mind more nodes more heap memory so you can control it from
> Here !!
> //BufferSize's got to be decreased due to latest refactorings in open62541 v1.2rc.
> UA_Int32 sendBufferSize = 10000; // 16384;
> UA_Int32 recvBufferSize = 10000; //16384;
> 
> and here
> xTaskCreatePinnedToCore(opcua_task, "opcua_task", 24000, NULL, 10, NULL, 0);
> 
> i will be happy if this solution helps the others 👍
> 
> greetings leon :)

Thank you very much.
I had the same problem. By the way, your answer was very helpful.!!!!!

leonvita91, 2021-06-25:
sure you welcome if you want to check more info about ESP32 POE with DHT22 and Bosch bme280 you can visit my profile :)